jeffsf/pyDE1 #2: try_de1 - flow_sequencer issue

g18xxgit, 2021-06-27:
One of the recent commits has broken examples/try_de1.py for me.  When I run it I get the following output:

> Traceback (most recent call last):
>   File "/home/username/Documents/tmp/pyDE1/examples/try_de1.py", line 213, in <module>
>     loop.run_until_complete(run(
>   File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
>     return future.result()
>   File "/home/username/Documents/tmp/pyDE1/examples/try_de1.py", line 62, in run
>     start_request_queue_processor(request_queue=request_queue,
> TypeError: start_request_queue_processor() missing 1 required positional argument: 'flow_sequencer'

I don't have any issues running examples/monitor_delay.py.

Steps to reproduce:

> $ git clone https://github.com/jeffsf/pyDE1.git
> $ cd pyDE1/examples/
> $ python try_de1.py

System information:
> $ python --version
> Python 3.9.5
> $ uname -a
> Linux hostname 5.10.42-1-MANJARO #1 SMP PREEMPT Thu Jun 3 14:37:11 UTC 2021 x86_64 GNU/Linux

I expect it's something trivial - any ideas?

jeffsf, 2021-06-27:
The FlowSequencer() was made a singleton in commit a420958b and shouldn't be required or being passed. from that commit oneard.

From a checkout at which commit are you seeing this ?

https://github.com/jeffsf/pyDE1/blob/a8b26866b89b9400a35d330d63120d7f5d0c5807/src/pyDE1/dispatcher/dispatcher.py#L72

g18xxgit, 2021-07-02:
Just got time to get back to this - thanks for your quick response.

Given your recent commits in the interim, I took a fresh cut of the source before addressing your comments above.  Luckily, the issue no longer presenting itself, so I'll close this ticket.

closed 2021-07-02 by g18xxgit