jeffsf/pyDE1 #4: [v0.5.0] [Low priority] Error on GET request to /de1 with no DE1 connected

g18xxgit, 2021-07-24:
After running  `src/pyDE1/run.py`, and then calling `curl -D - -X GET http://127.0.0.1:1234/de1` gives the following output from pyDE1:

> 2021-07-24 15:53:56,077 INFO [InboundAPI] InboundAPI: Request: GET /de1 HTTP/1.1
> 2021-07-24 15:53:56,078 DEBUG [Controller] Dispatcher: GET DE1 requires {'DE1': True, 'Scale': False}
> 2021-07-24 15:53:56,079 ERROR [Controller] Dispatcher: Exception in processing HTTPMethod.GET Resource.DE1 DE1NotConnectedError('DE1 not connected')
> 2021-07-24 15:53:56,080 ERROR [Controller] SupervisedTask._request_queue_processor: Task failed with exception:
> Traceback (most recent call last):
>   File "/home/shaitan/.local/lib/python3.9/site-packages/pyDE1/dispatcher/dispatcher.py", line 105, in _request_queue_processor
>     raise DE1NotConnectedError("DE1 not connected")
> pyDE1.exceptions.DE1NotConnectedError: DE1 not connected
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/home/shaitan/.local/lib/python3.9/site-packages/pyDE1/dispatcher/dispatcher.py", line 117, in _request_queue_processor
>     traceback.print_tb(tbe)
>   File "/usr/lib/python3.9/traceback.py", line 53, in print_tb
>     print_list(extract_tb(tb, limit=limit), file=file)
>   File "/usr/lib/python3.9/traceback.py", line 72, in extract_tb
>     return StackSummary.extract(walk_tb(tb), limit=limit)
>   File "/usr/lib/python3.9/traceback.py", line 347, in extract
>     for f, lineno in frame_gen:
>   File "/usr/lib/python3.9/traceback.py", line 312, in walk_tb
>     yield tb.tb_frame, tb.tb_lineno
> AttributeError: 'TracebackException' object has no attribute 'tb_frame'

From a client perspective, this is (unsurprisingly) the response seen to the curl command:

> $ curl -D - -X GET http://127.0.0.1:1234/de1
> HTTP/1.0 408 Request Timeout
> Server: BaseHTTP/0.6 Python/3.9.6
> Date: Sat, 24 Jul 2021 05:53:01 GMT
> Content-type: text/plain
> Content-length: 88
> Last-Modified: Sat, 24 Jul 2021 15:53:01 +1000
> 
> TimeoutError: Timeout waiting for response from controller, over 16.200000000000003 sec

Raising this as an issue given that, for example, the equivalent call to `http://.../scale` avoids the exception in pyDE1 and gives a more sensible response:

> $ curl -D - -X GET http://127.0.0.1:1234/scale
> HTTP/1.0 200 OK
> Server: BaseHTTP/0.6 Python/3.9.6
> Date: Sat, 24 Jul 2021 05:53:22 GMT
> Content-type: application/json
> Content-length: 228
> Last-Modified: Sat, 24 Jul 2021 15:53:22 +1000
> 
> {
>     "connectivity": {
>         "mode": null
>     },
>     "display": {
>         "display_on": null
>     },
>     "id": {
>         "id": null,
>         "name": null,
>         "type": null
>     },
>     "tare": {
>         "tare": null
>     }
> }

And similarly for a sub-component of the de1 branch all works as expected:

> $ curl -D - -X GET http://127.0.0.1:1234/de1/connectivity
> HTTP/1.0 200 OK
> Server: BaseHTTP/0.6 Python/3.9.6
> Date: Sat, 24 Jul 2021 05:53:48 GMT
> Content-type: application/json
> Content-length: 32
> Last-Modified: Sat, 24 Jul 2021 15:53:48 +1000
> 
> {
>     "mode": "not_connected"
> }

Let me know if you need additional information or this doesn't make sense.

jeffsf, 2021-07-24:
Looks like I missed making sure there was a traceback object present. I'll also have a "bail-out" if the traceback is missing in the future.

The 16-second timeout there is a secondary symptom.

You _should_ be getting an appropriate HTTP return code in these cases, along with a stack trace for debugging.

You also should _not_ be getting successful return codes when connectivity is needed to establish the data. You can tell, for example, _if_ the scale is connected, without it being connected. You should get a "409 Conflict" for these kind of things. I'll check through things. 

```
                if isinstance(resp.exception,
                              (DE1APIUnsupportedStateTransitionError,
                               DE1NotConnectedError,
                               DE1IsConnectedError,
                               DE1NoAddressError,
                               DE1OperationInProgressError,)):
                    http_status = HTTPStatus.CONFLICT
```

"Is connected" can happen if, for example, you try to change the address of an already connected device. You need to disconnect from the current address, then connect to the next address.

closed 2021-07-27 by jeffsf

jeffsf, 2021-07-27:
Thanks!

Found an error in branching as well as the wrong method to display the traceback.

commit 4bce9fd1 (2021-07-27): Correct reporting not-connected conditions to HTTP API

When the DE1 or scale is required to be connected to answer the query
but it is not, a 409 (Conflict) response should be returned.

In two cases, this was not happening. This resolves a timeout when
requesting from the DE1 when not connected, as well as a null response
involving a scale when one is not configured.

DE1NotConnectedError exceptions are now logged as INFO
as they generally shouldn't represent a coding or operational error

Fixes #4