jeffsf/pyDE1 #5: Should setting ID with first_if_found always try to connect?

jeffsf, 2021-07-25:
A user found the behavior surprising that setting the scale's ID (address) using `first_if_found` with a scale that had already been instantiated, though later disconnected, in the same session. There was an expectation that the operation would connect the now-disconnected scale.

```
2021-07-25 11:35:33,093 INFO [InboundAPI] InboundAPI: Request: PATCH /scale/id HTTP/1.1
2021-07-25 11:35:33,094 DEBUG [Controller] Dispatcher: PATCH SCALE_ID requires {'DE1': False, 'Scale': False}
2021-07-25 11:35:33,096 DEBUG [Controller] Scanner: Starting scanner
2021-07-25 11:35:33,097 INFO [OutboundAPI] OutboundAPI: {'ConnectivityChange': 1, 'ScannerNotification': 1}
2021-07-25 11:35:33,409 INFO [Controller] Scanner: 'Decent Scale' matched at FF:FF:10:aa:bb:cc by AdvertisementData(local_name='Decent Scale', manufacturer_data={513: b'\xff\xff\x10\x00\x02\xc0\x06d\x01\x10\x11'}, service_uuids=['0000fee7-0000-1000-8000-00805f9b34fb'])
2021-07-25 11:35:33,444 DEBUG [Controller] Scanner: Stopped scanner
2021-07-25 11:35:33,445 INFO [Controller] ScaleProcessor: Request to replace scale with FF:FF:10:aa:bb:cc
2021-07-25 11:35:33,445 INFO [Controller] ScaleProcessor: Already using FF:FF:10:aa:bb:cc. No action taken
2021-07-25 11:35:33,450 INFO [InboundAPI] InboundAPI: 127.0.0.1 - - [25/Jul/2021 11:35:33] "PATCH /scale/id HTTP/1.1" 200 -
2021-07-25 11:35:33,451 DEBUG [InboundAPI] InboundAPI: RTT: 358.4 ms PATCH /scale/id HTTP/1.1
```

The call resulted in the "technically correct" 200 result, as the scale is configured for the requested address.

Although the API call is to change the ID (address) of the scale, the behavior to connect to the scale _if it is new_  suggests that one might expect consistent "try to connect" with an instantiated, though disconnected scale.

Under consideration and evaluation at this time as:
* Adding an attempt to connect could add another 5 or more seconds to an already slow-to-return call
* The use case for the first_if_found was intended to be a convenience method for client start, after querying the scale
* Automated reconnect, yet to be implemented, may cause this use case to become overtaken by events

g18xxgit, 2021-07-25:
Another option (not saying it's better) might be that `first_if_found` merely scans and changes the address, so even on first time use it needs a follow-up call to `scale/connectivity/mode` to actually connect.  This may be implicitly clearer about the intent of the API?  It also limits the duration of the `first_if_found` call in case of timeout issues.

jeffsf, 2021-08-13:
At least for now, I'm going to say "no" as I haven't found any significant usability issues when developing a test-bed web app.

With a display in front of you if there is an address present, then you _connect,_ otherwise you _scan._ 

Feel free to reopen if you find usability issues with other implementations.

closed 2021-08-13 by jeffsf