jeffsf/pyDE1 #9: Can't Sleep from Refill state

jeffsf, 2021-12-05:
>  pyDE1 was refusing to switch from the state Refill to Sleep. I think this is useful so that you can turn it off if you don’t have the time/equipment/water to refill at the time.

Reported by AndrewB

jeffsf, 2021-12-05:
Will have to check to see if can Idle from Refill

Otherwise will need to rework https://github.com/jeffsf/pyDE1/blob/beta/src/pyDE1/de1/de1.py#L1441

jeffsf, 2021-12-05:
At least right now, it doesn't look as though the DE1 will always take action on an accepted Sleep request when in Refill. 

I'll keep looking into this to see if I can figure out the behavior.

a112358132134, 2021-12-07:
It appears this line in the official app's tcl indicates the "Exit App" button triggers an `{Exit}` command:
https://github.com/decentespresso/de1app/blob/1d5f65de8bb39d3cdd0ebf887ddc76a63e92a00e/de1plus/skins/default/standard_includes.tcl#L48

I don't understand half of what's going on in tcl, but this function looks suspicious:
https://github.com/decentespresso/de1app/blob/728c86575d9107d4d190bd9c617171eff4803a00/de1plus/bluetooth.tcl#L804
It shows from row 821 that there is a fail-over task to check the DE1 has gone to sleep; if not, the function it calls (just above in same file) appears to just stop running BLE scans, then stop the open BLE connection with the DE1.  Maybe this is how the app forces the sleep on exit?

jeffsf, 2021-12-10:
From what I can tell, while the DE1 will accept a Sleep request while in a refill (sub)state, it does not take action on it unless the state is cleared, such as with the GHC.

If you're working with a non-GHC machine, it may behave differently. With no visible indication of refill required, it may also be that the app simply disconnects. The code you identified appears to me to just be "normal housekeeping" of closing open Bluetooth handles on app exit. As de1app is just a Tcl script and Androwish is the app that is working with the Android Bluetooth stack, it may or may not explicitly close the handles on script termination.

I'll add the ability to at least request Sleep during refill in the near future.

commit 55d81bbc (2021-12-10): Allow DE1 Sleep request from Refill state

Should be able to defer the refill until a later time and sleep the DE1.
This will allow a Sleep request to be sent. However, it appears that
the DE1 ignores it, at least until interrupted by the GHC.

Closes #9
closed 2021-12-10 by jeffsf