jeffsf/pyDE1 #16: [v1.0.0] [Low priority] Malformed http request to PATCH `stop_at_weight` not cleanly handled

a112358132134, 2022-01-10:
Whilst using curl, I accidentally added quotations to the double value against `stop_at_weight`:
```
curl -D - -X PATCH --data '{"stop_at_weight": "54.0"}' http://192.168.0.24:1234/de1/control/espresso
```
This gives the odd response of:
```
curl: (52) Empty reply from server
```
`journalctl` shows the following:
```
Jan 11 09:26:57 de1server pyde1[8931]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/api/inbound/http/run.py", line 161, in __init__
Jan 11 09:26:57 de1server pyde1[8931]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/api/inbound/http/run.py", line 448, in do_PATCH
Jan 11 09:26:57 de1server pyde1[8931]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/dispatcher/validate.py", line 62, in validate_patch_return_targets
Jan 11 09:26:57 de1server pyde1[8931]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/dispatcher/validate.py", line 120, in _validate_patch_inner
```

I think this malformed request should return a response with the error, if it follows the same approach as other invalid requests?

jeffsf, 2022-01-11:
Replicated. Looks to be a coding error on handling the failure to receive the proper type.

Looks like your distro's utilities are just a bit different and may not handle multi-line entries as well.
```
Jan 10 17:08:25 pi-walnut pyde1[29169]: ----------------------------------------
Jan 10 17:08:25 pi-walnut pyde1[29169]: Exception occurred during processing of request from ('127.0.0.1', 55832)
Jan 10 17:08:25 pi-walnut pyde1[29169]: Traceback (most recent call last):
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/usr/lib/python3.9/socketserver.py", line 316, in _handle_request_noblock
Jan 10 17:08:25 pi-walnut pyde1[29169]:     self.process_request(request, client_address)
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/usr/lib/python3.9/socketserver.py", line 347, in process_request
Jan 10 17:08:25 pi-walnut pyde1[29169]:     self.finish_request(request, client_address)
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/usr/lib/python3.9/socketserver.py", line 360, in finish_request
Jan 10 17:08:25 pi-walnut pyde1[29169]:     self.RequestHandlerClass(request, client_address, self)
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/api/inbound/http/run.py", line 161, in __init__
Jan 10 17:08:25 pi-walnut pyde1[29169]:     super(RequestHandler, self).__init__(*args, **kwargs)
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/usr/lib/python3.9/socketserver.py", line 720, in __init__
Jan 10 17:08:25 pi-walnut pyde1[29169]:     self.handle()
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/usr/lib/python3.9/http/server.py", line 427, in handle
Jan 10 17:08:25 pi-walnut pyde1[29169]:     self.handle_one_request()
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/usr/lib/python3.9/http/server.py", line 415, in handle_one_request
Jan 10 17:08:25 pi-walnut pyde1[29169]:     method()
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/api/inbound/http/run.py", line 448, in do_PATCH
Jan 10 17:08:25 pi-walnut pyde1[29169]:     targets = validate_patch_return_targets(resource=resource,
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/dispatcher/validate.py", line 62, in validate_patch_return_targets
Jan 10 17:08:25 pi-walnut pyde1[29169]:     _validate_patch_inner(patch=patch,
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/home/pyde1/venv/pyde1/lib/python3.9/site-packages/pyDE1/dispatcher/validate.py", line 120, in _validate_patch_inner
Jan 10 17:08:25 pi-walnut pyde1[29169]:     f"Expected {mapping_value.v_type.__name__} value "
Jan 10 17:08:25 pi-walnut pyde1[29169]:   File "/usr/lib/python3.9/typing.py", line 694, in __getattr__
Jan 10 17:08:25 pi-walnut pyde1[29169]:     raise AttributeError(attr)
Jan 10 17:08:25 pi-walnut pyde1[29169]: AttributeError: __name__
Jan 10 17:08:25 pi-walnut pyde1[29169]: ----------------------------------------

```

jeffsf, 2022-01-11:
Resolution will provide
```
$ curl -D - -X PATCH --data '{"stop_at_weight": "54.0"}' http://localhost:1234/de1/control/espresso
HTTP/1.0 400 Bad Request
Server: BaseHTTP/0.6 Python/3.9.2
Date: Tue, 11 Jan 2022 01:52:59 GMT
Content-type: text/plain
Content-length: 98
Last-Modified: Mon, 10 Jan 2022 17:52:59 -0800

DE1APITypeError("Expected typing.Optional[float] value at stop_at_weight, not 54.0 <class 'str'>")
```

commit 43614df5 (2022-01-11): HTTP API: Type checker handles errors matching Optional[type]

If the v_type was not a simple type, it did not have a __name__
attribute, causing an exception when trying to report the error.

Also enhanced the message to make it clear that 54.0 was interpreted
as a string, not a float

DE1APITypeError(
    "Expected typing.Optional[float] value at stop_at_weight,
     not 54.0 <class 'str'>"
)

Closes #16
closed 2022-01-24 by jeffsf