jeffsf/pyDE1 #18: [v1.0.0] Duplicate mqtt SequencerGateNotification streams on flush?

a112358132134, 2022-01-11:
I've seen the following strange output from my app's backed when debugging my handling of a `SequencerGateNotification` for a flush.  It indicates that two complete sequences with the same ID have been emitted sequentially for a single flush.

Is this expected behavior?  My rough understanding is that each of the GHC Actions (e.g. flush, espresso) should have a single sequence ID for each execution.

```
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6087787, "create_time": 1641939252.6089156, "name": "sequence_start", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6524212, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> New sequence commencing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2 (was: )
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6151524, "create_time": 1641939252.615221, "name": "sequence_flow_begin", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6536167, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6168492, "create_time": 1641939252.616894, "name": "sequence_expect_drops", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.654736, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6182375, "create_time": 1641939252.6182697, "name": "sequence_exit_preinfuse", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6560616, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6199021, "create_time": 1641939252.6199794, "name": "sequence_flow_end", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6571827, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6225045, "create_time": 1641939252.6225994, "name": "sequence_flow_state_exit", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6582475, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6250043, "create_time": 1641939252.6250787, "name": "sequence_last_drops", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6596482, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.626708, "create_time": 1641939252.6267545, "name": "sequence_complete", "action": "clear", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6611829, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.6419997, "create_time": 1641939252.6420748, "name": "sequence_start", "action": "set", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.6962492, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.644863, "create_time": 1641939252.644923, "name": "sequence_flow_begin", "action": "set", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.7035704, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939252.7298067, "create_time": 1641939252.7298834, "name": "sequence_expect_drops", "action": "set", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939252.7460005, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939255.669636, "create_time": 1641939255.669694, "name": "sequence_flow_end", "action": "set", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939255.7062993, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Packet received: {"arrival_time": 1641939255.6718736, "create_time": 1641939255.6719303, "name": "sequence_flow_state_exit", "action": "set", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939255.7076762, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa17fa640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa0d48640> Packet received: {"arrival_time": 1641939255.6891205, "create_time": 1641939255.6892154, "name": "sequence_last_drops", "action": "set", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939255.7095644, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fff7bfff640> Packet received: {"arrival_time": 1641939255.6949747, "create_time": 1641939255.695503, "name": "sequence_complete", "action": "set", "sequence_id": "0117351f-d2ac-4e8c-836a-0df5c7d61db2", "active_state": "HotWaterRinse", "version": "1.1.0", "event_time": 1641939255.7113047, "sender": "FlowSequencerImpl", "class": "SequencerGateNotification"}
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fffa0d48640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
DEBUG [SequencerGateNotificationProcessor] <Thread 0x7fff7bfff640> Same sequence continuing: 0117351f-d2ac-4e8c-836a-0df5c7d61db2
```

jeffsf, 2022-01-11:
Working as expected
`"action": "clear"` and `"action": "set"`

closed 2022-01-11 by jeffsf

a112358132134, 2022-01-13:
Thanks for picking that up, sorry for the effort on such a silly mistake.  Bug was in my filtering code that was already supposed to have cut `"clear"` messages out.