jeffsf/pyDE1 #19: [v1.0.0] Request for independent test of scale disconnection behaviour

a112358132134, 2022-01-13:
I have an odd scenario occurring that I'm hoping someone can independently reconfirm or disprove.

Steps to reperform:
1. Start mqtt broker, pyde1
2. Connect to DE1, scale
3. Forcefully switch scale off (e.g. pull out power)

Behaviour observed after step 3:
 - As expected, after a short wait (1-3 seconds), pyde1 reports scale as disconnected.
 - However, shortly after that (1-3 seconds) pyde1 also reports the DE1 as disconnected.  Note that a reconnection is quickly attempted and subsequently succeeds, so this is a temporary situation.

The odd behaviour (from my perspective) is the impact the scale disconnection appears to have on the DE1 connection.  Does anyone else see this, or is it something unique to my setup?  If not unique to me, is this supposed to happen?

NB: Nothing appears in `journalctl` about any of this.

jeffsf, 2022-01-13:
You can enable low-level Bluetooth debugging by changing the `LOGFILE` handler the `root.bleak` logger to `DEBUG`

This is likely an OS or hardware issue. 

Is this with the Decent Scale, which is unsupported?

jeffsf, 2022-01-13:
```
2022-01-13 08:34:08,872 INFO [Controller] Scale: disconnect_callback: Disconnected from AtomaxSkaleII at FF:06:AF:aa:bb:cc, willful_disconnect: False
2022-01-13 08:34:08,889 INFO [Controller] Scale: Will try reconnecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:08,890 INFO [Controller] Scale: Connecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:19,113 ERROR [Controller] Scale: Connection failed to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:19,122 INFO [Controller] Scale: Will try reconnecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:19,123 INFO [Controller] Scale: Connecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:29,303 ERROR [Controller] Scale: Connection failed to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:29,315 INFO [Controller] Scale: Will try reconnecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:29,316 INFO [Controller] Scale: Connecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:39,487 ERROR [Controller] Scale: Connection failed to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:39,491 INFO [Controller] Scale: Will try reconnecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:39,492 INFO [Controller] Scale: Connecting to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:46,565 INFO [Controller] Scale: Connected to AtomaxSkaleII at FF:06:AF:aa:bb:cc
2022-01-13 08:34:46,571 INFO [Controller] Scale: Scale.standard_initialization()
2022-01-13 08:34:46,832 INFO [Controller] Scale.AtomaxIIScale: Display on
2022-01-13 08:34:46,920 INFO [Controller] Scale.AtomaxIIScale: Sending weight updates
2022-01-13 08:34:46,929 WARNING [Controller] Scale.Processor: Resetting scale due to gap in reports: 38.579 > 0.8 s
2022-01-13 08:34:47,035 INFO [Controller] Scale.AtomaxIIScale: Sending button updates
2022-01-13 08:34:47,063 INFO [Controller] Scale: Loading scale-period estimate of 0.10312 from database
2022-01-13 08:34:47,157 INFO [Controller] Scale.AtomaxIIScale: Grams selected
2022-01-13 08:34:47,160 INFO [Controller] Scale: Ready
```

a112358132134, 2022-01-13:
Thanks @jeffsf.

Yes, this is with the Decent Scale, which I appreciate is unsupported.

With the DEBUG flag set for bleak, there are no smoking guns on a root cause.  I also tried the other way around (i.e. switching off the DE1), and got the same result (scale briefly disconnects before reconnecting).

All my systems are running bluez v5.62 which, from a bit of searching, appears to have a lot of anecdotally reported connection regressions over prior versions.  The only bluetooth devices I use are the DE1 and the Decent Scale, so I don't have a broader perspective on the validity of these opinions.  I assume you're running the default bullseye bluez v5.55 on your rpi?

Here is the full log output for a forced scale disconnect:
```
2022-01-14 08:52:30,021 INFO [MainProcess] Config.YAML: Loading config overrides from /usr/local/etc/pyde1/pyde1.conf
2022-01-14 08:52:30,128 DEBUG [MainProcess] Config.YAML: Setting mqtt.TOPIC_ROOT
2022-01-14 08:52:30,129 DEBUG [MainProcess] Config.YAML: Setting mqtt.CLIENT_ID_PREFIX
2022-01-14 08:52:30,130 DEBUG [MainProcess] Config.YAML: Setting mqtt.BROKER_HOSTNAME
2022-01-14 08:52:30,131 DEBUG [MainProcess] Config.YAML: Setting mqtt.BROKER_PORT
2022-01-14 08:52:30,132 DEBUG [MainProcess] Config.YAML: Setting mqtt.TRANSPORT
2022-01-14 08:52:30,133 DEBUG [MainProcess] Config.YAML: Setting mqtt.KEEPALIVE
2022-01-14 08:52:30,134 DEBUG [MainProcess] Config.YAML: Setting mqtt.USERNAME
2022-01-14 08:52:30,135 DEBUG [MainProcess] Config.YAML: Setting mqtt.PASSWORD
2022-01-14 08:52:30,136 DEBUG [MainProcess] Config.YAML: Setting mqtt.DEBUG
2022-01-14 08:52:30,137 DEBUG [MainProcess] Config.YAML: Setting http.SERVER_HOST
2022-01-14 08:52:30,138 DEBUG [MainProcess] Config.YAML: Setting http.SERVER_PORT
2022-01-14 08:52:30,139 DEBUG [MainProcess] Config.YAML: Setting http.SERVER_ROOT
2022-01-14 08:52:30,140 DEBUG [MainProcess] Config.YAML: Setting logging.LOG_DIRECTORY
2022-01-14 08:52:30,141 DEBUG [MainProcess] Config.YAML: Setting logging.LOG_FILENAME
2022-01-14 08:52:30,141 DEBUG [MainProcess] Config.YAML: Setting logging.formatters.STYLE
2022-01-14 08:52:30,142 DEBUG [MainProcess] Config.YAML: Setting logging.formatters.LOGFILE
2022-01-14 08:52:30,143 DEBUG [MainProcess] Config.YAML: Setting logging.formatters.MQTT
2022-01-14 08:52:30,144 DEBUG [MainProcess] Config.YAML: Setting logging.formatters.STDERR
2022-01-14 08:52:30,145 DEBUG [MainProcess] Config.YAML: Setting logging.handlers.LOGFILE
2022-01-14 08:52:30,146 DEBUG [MainProcess] Config.YAML: Setting logging.handlers.MQTT
2022-01-14 08:52:30,147 DEBUG [MainProcess] Config.YAML: Setting logging.handlers.STDERR
2022-01-14 08:52:30,148 DEBUG [MainProcess] Config.YAML: Setting logging.LOGGERS
2022-01-14 08:52:30,149 DEBUG [MainProcess] Config.YAML: Setting bluetooth.SCAN_TIME
2022-01-14 08:52:30,150 DEBUG [MainProcess] Config.YAML: Setting bluetooth.CONNECT_TIMEOUT
2022-01-14 08:52:30,151 DEBUG [MainProcess] Config.YAML: Setting bluetooth.DISCONNECT_TIMEOUT
2022-01-14 08:52:30,151 DEBUG [MainProcess] Config.YAML: Setting bluetooth.SCAN_CACHE_EXPIRY
2022-01-14 08:52:30,152 DEBUG [MainProcess] Config.YAML: Setting bluetooth.RECONNECT_RETRY_COUNT
2022-01-14 08:52:30,153 DEBUG [MainProcess] Config.YAML: Setting bluetooth.RECONNECT_GAP
2022-01-14 08:52:30,155 DEBUG [MainProcess] Config.YAML: Setting bluetooth.ID_FILE_DIRECTORY
2022-01-14 08:52:30,156 DEBUG [MainProcess] Config.YAML: Setting bluetooth.ID_FILE_SUFFIX
2022-01-14 08:52:30,157 DEBUG [MainProcess] Config.YAML: Setting database.FILENAME
2022-01-14 08:52:30,158 DEBUG [MainProcess] Config.YAML: Setting de1.LINE_FREQUENCY
2022-01-14 08:52:30,159 DEBUG [MainProcess] Config.YAML: Setting de1.STOP_AT_WEIGHT_ADJUST
2022-01-14 08:52:30,160 INFO [MainProcess] Config.YAML: Config overrides loaded from /usr/local/etc/pyde1/pyde1.conf
2022-01-14 08:52:30,813 INFO [MainProcess] root: Configured stderr_handler: <StreamHandler <stderr> (WARNING)>
2022-01-14 08:52:30,815 INFO [MainProcess] root: Configured mqtt_handler: <PipeHandler (ERROR)>
2022-01-14 08:52:30,823 INFO [MainProcess] root: Configured logfile_handler: <WatchedFileHandler /var/log/pyde1/pyde1.log (INFO)>
2022-01-14 08:52:30,825 INFO [MainProcess] root: Started <logging.handlers.QueueListener object at 0xffff97955a60>
2022-01-14 08:52:30,827 DEBUG [MainProcess] root: log_queue_listener handlers: (<StreamHandler <stderr> (WARNING)>, <PipeHandler (ERROR)>, <WatchedFileHandler /var/log/pyde1/pyde1.log (INFO)>)
2022-01-14 08:52:30,874 INFO [MainProcess] Logging.Config: Setting root.aiosqlite to INFO
2022-01-14 08:52:30,877 INFO [MainProcess] Logging.Config: Setting root.asyncio to INFO
2022-01-14 08:52:30,878 INFO [MainProcess] Logging.Config: Setting root.bleak to DEBUG
2022-01-14 08:52:30,933 INFO [MainProcess] Supervised.Process.LogMQTT: Starting
2022-01-14 08:52:30,956 INFO [MainProcess] Supervised.Process.LogMQTT: Started: <Process name='LogMQTT' pid=3283 parent=3237 started>
2022-01-14 08:52:30,961 INFO [MainProcess] Supervised.Process.OutboundAPI: Starting
2022-01-14 08:52:30,981 INFO [MainProcess] Supervised.Process.OutboundAPI: Started: <Process name='OutboundAPI' pid=3284 parent=3237 started>
2022-01-14 08:52:30,987 INFO [MainProcess] Supervised.Process.InboundAPI: Starting
2022-01-14 08:52:31,011 INFO [MainProcess] Supervised.Process.InboundAPI: Started: <Process name='InboundAPI' pid=3285 parent=3237 started>
2022-01-14 08:52:31,031 INFO [MainProcess] Supervised.Process.DatabaseLogger: Starting
[... cut ...]
2022-01-14 08:54:56,955 INFO [Controller] root.Scale.DecentScale: Sending weight updates
2022-01-14 08:54:56,956 INFO [Controller] root.Scale.DecentScale: Sending button updates
2022-01-14 08:54:56,981 INFO [Controller] Scale: Loading scale-period estimate of 0.10528 from database
2022-01-14 08:54:56,990 INFO [Controller] Scale: Ready
2022-01-14 08:57:08,264 WARNING [Controller] root.asyncio: Executing <Handle MessageBus._message_reader() created at /usr/lib/python3.9/asyncio/selector_events.py:259> took 0.272 seconds
```

a112358132134, 2022-01-13:
There are a few DBus-related connection errors in bluez 5.62 that are apparently fixed in 5.63, including:
https://github.com/bluez/bluez/commit/ebf2d7935690c00c7fd12768177e2023fc63c9fe

5.63 was released ~5 days ago, hence should hit Manjaro stable shortly.  I'll retest once it's available.

closed 2022-01-14 by a112358132134

jeffsf, 2022-01-14:
> running the default bullseye bluez v5.55

Yes, v5.55

I'll avoid upgrading until at least 5.63, from the sound of it!