jeffsf/pyDE1 #21: Feature request: Additional calibration options

a112358132134, 2022-01-14:
On the official app, there are two more calibration options available than via the pyDE1 http API:
- Temperature offset
- Pressure offset

As a low priority, could I please request these are included in the API?

I have used external equipment to adjust the calibration (slightly) differently to the factory calibration values for these items in the official app, and would like the ability to continue to do so.

jeffsf, 2022-01-14:
I was asked not to publish information on the WriteKey.

All are already _read_ 

```
2022-01-13 18:10:16,276 INFO [Controller] DE1.CUUID.Calibration.Notify: Read CalFlow de1: 1.0 meas: 1.10001
2022-01-13 18:10:16,905 INFO [Controller] DE1.CUUID.Calibration.Notify: ReadFactory CalFlow de1: 1.0 meas: 1.10001
2022-01-13 18:10:17,537 INFO [Controller] DE1.CUUID.Calibration.Notify: Read CalPressure de1: 1.0 meas: 1.13420
2022-01-13 18:10:18,137 INFO [Controller] DE1.CUUID.Calibration.Notify: ReadFactory CalPressure de1: 1.0 meas: 1.13420
2022-01-13 18:10:18,772 INFO [Controller] DE1.CUUID.Calibration.Notify: Read CalTemp de1: 0.0 meas: -3.0
2022-01-13 18:10:19,407 INFO [Controller] DE1.CUUID.Calibration.Notify: ReadFactory CalTemp de1: 0.0 meas: -3.0
```

a112358132134, 2022-01-14:
So given this limitation, can you please confirm I understand the situation?

1. Any calibration changes to these items will still need to be done in the official app
2. Those calibrations will persist with use of pyde1

jeffsf, 2022-01-14:
The calibration values are persisted in the DE1

To write to the calibration registers in the DE1, you need the appropriate WriteKey

The calls are in place in pyDE1 to both decode and encode the Calibration PackedAttribute. See `pyDE1/de1/c_api.py`

a112358132134, 2022-01-16:
Ok, thanks @jeffsf, appreciate the clarification.

closed 2022-01-16 by a112358132134