jeffsf/pyDE1 #25: Can't "lose" a scale

jeffsf, 2022-01-27:
Disconnect retains the scale's MAC address, making it difficult to _change_ scales

jeffsf, 2022-02-02:
Current disconnect behavior is inconsistent:

* DE1 wipes the address (because "the only reason to disconnect would be to _change_ DE1s")
* Scale retains the address (because, well, I don't recall)

Neither of these assumptions or behaviors is right all the time for all people and use cases.

People are potentially disconnecting a DE1 to use it with different control software.

It might be the case that people are wanting to switch scales (without restarting pyDE1), use the scale with different software (pour-over, for example) and, potentially using a second scale for beans/grinds.

jeffsf, 2022-02-06:
Working toward adopting capture/release developed for https://github.com/jeffsf/steam-to-temperature

Not _quite_ a drop-in, unfortunately

https://github.com/jeffsf/pyDE1/commit/207a492bcff66b18a4f638e2dc0dc54dce22354a relates to this effort

a112358132134, 2022-02-12:
Another consideration, at least with bluez, is that it does not appear (in my practical use) to like making any connection if it considers an ID to be "stale".  I don't know whether this is the same on other stacks/platforms.

In this context "stale" would mean has been too long since a scan was performed, or too long since disconnection of the device.   I don't know exactly how long this timeframe is, but it seems it could easily be under a minute.

In the GUI layer, I've taken to forcing a scan before all new connections are requested as a result, because the failure rate was otherwise extremely high.

jeffsf, 2022-02-12:
Filed as https://github.com/hbldh/bleak/issues/713 

The "experimental" flag that they use to try to get around a scan isn't standard, and isn't intended for general use, as per the Bluez mailing lists.

Workaround right now is
```
                            self._bleak_client._device_path = None
                            self._bleak_client._device_info = None
```

jeffsf, 2022-10-18:
As of bleak v0.18.1 (and perhaps earlier), clearing `_device_path` should be sufficient

```
        except bleak.exc.BleakDBusError as e:
            e: bleak.exc.BleakDBusError
            if (e.dbus_error == 'org.freedesktop.DBus.Error.UnknownObject'
                    and e.dbus_error_details.startswith(
                        'Method "Connect" with signature "" on interface')
                    and e.dbus_error_details.endswith("doesn't exist\n")):
                self.logger.info("Seemingly stale device, resetting.")
                self._backend._device_path = None
```

closed 2023-02-13 by jeffsf