jeffsf/pyDE1 #27: Use of deprecated Callable definition

a112358132134, 2022-02-10:
When running the release version `1.1.0` of pyde1 against python 3.10, pyde1 crashes shortly after startup.  The root cause appears to be use of a deprecated importation definition of `Callable`.

With the following edit to line 51 of `src/pyDE1/api/outbound/mqtt/run.py` this error disappears:
```
# OLD line 51
from collections import Callable
# NEW line 51
from collections.abc import Callable
```

Relevant documentation highlighting this was deprecated in python 3.3, and removed in 3.10:
[https://docs.python.org/3.9/library/collections.html](https://docs.python.org/3.9/library/collections.html)
```
Deprecated since version 3.3, will be removed in version 3.10: Moved [Collections Abstract Base Classes](https://docs.python.org/3.9/library/collections.abc.html#collections-abstract-base-classes) to the [collections.abc](https://docs.python.org/3.9/library/collections.abc.html#module-collections.abc) module. For backwards compatibility, they continue to be visible in this module through Python 3.9.
```

jeffsf, 2022-02-10:
Thanks!

I didn't find any other "inappropriate" uses of `collections` as `collections.deque` still appears [there in Python 3.10](https://docs.python.org/3.10/library/collections.html)

If I missed one, please let me know.

a112358132134, 2022-02-12:
Looks like none of the actual containers moved to `.abc`, just the base classes that underpin/support them.

No other issues have come up for me using 3.10 apart from the one noted above. 

Thanks for resolving.

commit 39f7a576 (2022-02-10): Change import for Callable from collections to typing

Callable in collections has been deprecated and removed in Python 3.10
Change to 'from typing import Callable'

Closes #27
closed 2022-06-21 by jeffsf