jeffsf/pyDE1 #37: Shot replays?

a112358132134, 2022-12-28:
A query, not an issue.

In trying to implement a new shot graph display (in a more performant and feature rich library), I've run into a speed bump which is slowing down testing.  Can you please advise if there is an existing API or expectation on how these would be addressed?  I'd prefer not to "roll my own" unless it's necessary.  Thank you!

### Shot replays

When testing the UX, it would be extremely helpful to be able to trigger a replay of a prior shot (or flush, steam, or water pour).  The intent here is that the GUI app would be presented information that was consistent for a live shot, via the standard interface in MQTT.

Even better would be if there was some kind of flag set to show it's a replay of shot x rather than a live data feed, but I think this is nice to have.

jeffsf, 2022-12-29:
I think you're looking for https://github.com/jeffsf/pyDE1/blob/stable/src/pyDE1/services/runnable/replay.py

I haven't updated it for the new availability packets or the thermometer packets. If you need those sooner than later, let me know and I'll make the changes with the next Alpha drop.

a112358132134, 2022-12-29:
Perfect, thank you!

No problem re: new packets - hopefully I'll be able to test those few separately in real time.

a112358132134, 2022-12-29:
Query resolved - replay.py

closed 2022-12-29 by a112358132134