jeffsf/pyDE1 #40: PYDE1_ROOT in 40-config-files.sh does not resolve

FIRKremer, 2023-03-24:
Install of pyDE1 as per documentation fails at 40-config-files.sh, as directories rely on a PYDE1_ROOT  variable that is not set.
It is required to move services from the venv to /usr/local/etc/pyde1, but I am also not able to locate respective services in the VENV_PATH set by _config.

a112358132134, 2023-03-25:
That problem looks to have been created in a recent commit, where the definition for `PYDE1_ROOT` was removed:
https://github.com/jeffsf/pyDE1/commit/6b32f34095c4add32e0b9b45d9f05c4d7b31cd44

I don't have an understanding of the python venv stuff to fix it properly (jeffsf can help here.

In the interim, you could try hard coding the directory in the script.  By default, I think `PYDE1_ROOT` is set as `/home/pyde1/venv/pyde1/lib/python3.10/site-packages/pyDE1`?

jeffsf, 2023-03-28:
My apologies for not seeing this sooner. We've been without Internet since the storm a week ago and just got it back.

I should re-examine all of these, especially with the changes in v2.0 in how the executables are deployed by pip.